geeksville/Micro-RTSP #1: Support resolutions higher than SVGA

geeksville, 2019-02-03:
From jii at hackaday:

> So what exactly limits this to SVGA? The datasheet linked below says 1632×1232.

My comment:

ok - I poked around and figured out why the espressif example code used SVGA res.  The size needed for the framebuffer grows too big for the 512KB ram on the ESP32 (and the three boards I've tried don't have the ESP-PSRAM32 chip - which their camera driver _does_ support).  

I'm just using their traditional DMA by line a (double) framebuffer.  With the current code there is 255KB free after the two frame buffers (and the various DMA transfer buffers) are allocated.  But if I bump up to the larger frame size, I can't allocate the second framebuffer.

That said, processing the images from the camera was mostly just made easier by having the frame buffer memory.  Since the frames are being fired off via UDP in fragments about 1100 bytes at a time, it would be possible to just use the i2s layer directly (rather than the espressif camera driver) and as enough bytes arrive fire them off immediately.  This would _dramatically_ lower the memory needs (and allow full res use of the imager).

Alas though there are a few things in my queue (other hw projects) before I can do such surgery on this lib.  I'm recording this as an issue in case someone else can get to it first.  I'm happy to provide pointers / accept pull requests if anyone else wants to do this soon.

geeksville, 2019-02-03:
implementation plan:
Provide an alternate implementation of dma_finish_frame and dma_filter_buffer here: https://github.com/espressif/esp32-camera/blob/10b3d3c2a94ab44c92c199946190bbb4bbab5a95/driver/camera.c#L680

And instead of copying bytes to the frame buffer in filter_buffer, invoke a "bytesArrived" callback provided by the RTSP code.  Also change camera init to not allocate a framebuffer.

Niek, 2019-03-26:
+1 for this, also setting this value to 10000000 will double the framerate: https://github.com/geeksville/Micro-RTSP/blob/c0b1a1cc5d2465ac77b8caf0ba3c522b0607880f/src/OV2640.cpp#L27

geeksville, 2019-03-26:
ooh - great find!  Alas, I haven't gotten back to this (and probably won't for a while)  mainly because the WROVER versions of the boards have PSRAM that makes this possible without doing this extra work.  Also, I've been working on extending this into something more generally useful: https://github.com/geeksville/ezdevice-esp32