geeksville/Micro-RTSP #2: esp_camera.h: No such file or director

lazyzero, 2019-02-04:
Hi,

I try your TenDollarWebcam code and it gives me the following error message. 

> <command-line>:0:16: warning: ISO C++11 requires whitespace after the macro name
> In file included from ~/Projects/ESP_Projekte/TenDollarWebcam/src/ESP32-devcam.ino:1:0:
> .piolibdeps/Micro-RTSP/src/OV2640.h:9:24: fatal error: esp_camera.h: No such file or directory

This seams to be an unsatisfied dependency to https://github.com/espressif/esp32-camera how can I solve it?

Cheers Christian

geeksville, 2019-02-04:
hmm - It sounds like in the instructions I might have missed a "import project" step into platform IO.  can you use the platform IO and "import an existing project" and point it at the tendollarwebcam directory?  It sounds like it didn't auto fetch the espressio platform dependency in this file:

https://github.com/geeksville/TenDollarWebcam/blob/master/platformio.ini

lazyzero, 2019-02-05:
I solved the issue. It was due to the fact that I had a very old version of espressif32 installed. I suggest to use the line 
`platform = espressif32^1.6.0`
or `platform = espressif32>=1.6.0`

to make sure a compatible platform is loaded for the project.

Thanks Christian

closed 2019-02-05 by geeksville