geeksville/Micro-RTSP #4: GPIO mappings for this particular board type broken?

geeksville, 2019-02-06:
Someone on hackster claims this dev doesn't work anymore: https://www.banggood.com/Geekcreit-ESP32-CAM-WiFi-+-Bluetooth-Camera-Module-Development-Board-ESP32-With-Camera-Module-OV2640--p-1394679.html?rmmds=myorder&cur_warehouse=CN 

Possibly I messed up the gpio config structure.  Need to check their datasheet.

geeksville, 2019-02-08:
ah yes - I think the pin mappings for this board are different (oops): https://github.com/donny681/ESP32_CAMERA_QR#connect 

And I remembered why I stopped using it (even though it has PSRAM) - I smoked my USB to serial adapter.  After my replacement arrives (monday) I'll check that these mappings work and push a fix.

geeksville, 2019-02-10:
Fixed in https://github.com/geeksville/Micro-RTSP/commit/53284101a2442a954695ef0a27e00e45d2d7a9ce

bbbenji, 2019-02-11:
That fixed it! Thanks @geeksville

closed 2019-03-12 by geeksville