geeksville/Micro-RTSP #9: Video stops after some seconds

fedyfausto, 2019-04-07:
Hello there! I'm trying to use rtsp example with VLC (suing rtsp://ip:port/mjepg/1 but after some seconds the stream stops and in seral i ever get the warning message that says the frame is slow by some ms (average 100 ms)

bartlomiejcieszkowski, 2019-09-15:
Have you tried lowering the resolution of a camera?
See OV2640.cpp
and try changing frame_size to FRAMESIZE_QVGA, it should help.

rikki78, 2019-11-19:
I have the same issue. Changing the resolution didn't help.
I also got UDP errors the first time it connects.

soovui, 2020-04-22:
I have the same issue after some minutes it hangs and need to do manual reset.  I have tested a few rtsp from others too.  Giving the same problem.  Can't find any extremely stable rtsp for esp32 can.  Doing all sorts of trial like changing to a better power supply, also won't help. Could be the rtsp protocol not stablenor cannot handle well by esp32 cam.