geeksville/Micro-RTSP #12: AIThinker ESP32-CAM

ameennihad, 2019-05-31:
Hi,
I uploaded the code to AIThinker ESP32-CAM board, I'm getting following messages on serial monitor:

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

The ESP32 Camera example sketch works fine with my board when I select CAMERA_MODEL_AI_THINKER.

Is this the camera used with AIThinker board supported by the library? if yes, what do i need to do to use it?

Thanks

Ameen

ameennihad, 2019-05-31:
I found the solution, in the Setup I just changed:
cam.init(esp32cam_config);
to:
cam.init(esp32cam_aithinker_config);

geeksville, 2019-06-24:
yep - that's required - the aithinker board needs different setting.

closed 2019-06-24 by geeksville