geeksville/Micro-RTSP #13: Have some seconds delay.

mikelcq, 2019-06-05:
Hi Kevin,

Nice work! 
I had a chance to add this to ESP-Who project, just testing for the CAM streaming via UDP.  I put the rstp request reading (on rstp port ) in a separate (freertos)task, and VLC plays well. One problem is that there's a 2~3 seconds delay, but the view is clear and smooth.  I checked the overall time for one frame sending (actually rtsp.broadcastCurrentFrame()), about less than 100ms (about 60~80ms as I remember) in 640x480 res.  I'm figuring out what's the root cause, because it should be no delay and better than the httpserver's TCP streaming example(in ESP-Who),which has no delay.   Appreciate any help, thanks!

Mike

geeksville, 2019-06-24:
hmm - I'm not sure what the latency is in the camera chip (which does the JPEG compression), but I think about 300ms.  That wouldn't explain 2 secs though.

bartlomiejcieszkowski, 2020-01-04:
after doing some digging i found why there was a delay

it wasn't caused by the esp - cause let's think for a second - If the processing would be consuming too much time - we would see dropped frames, but with the delay - where would the esp store 2-3 second of a buffer ? ;) 

it was .. VLC
[image]

change the value to 0 and it should help
(also the VLC doesn't remember changes to cache, so if we select address from a list - it will have value set to a default 1000 -_- )

geeksville, 2020-01-05:
Great info!  If you have time and are willing could you add this note to
readme.md?  Others have asked about it.  I'd eagerly accept a pull request.

(Sent from a phone - please ignore typos)

Net-time, 2020-02-09:
You can change the Vlc cache permanently in advanced settings, usually you should not go below 300 unless you have a very stable network.

DoinStuff35, 2020-04-15:
Thanks for the Info! Did help a lot. But still have a ~1sec. delay on RTSP via VLC with 0ms Caching. HTTP works fine, no delay.

Any Ideas?

Net-time, 2020-04-16:
First of all the HTTP version is not actually a stream it is separate single jpeg images so quite easy to encode and send. RTSP is another beast and not sure if the ESP is capable of shortening the delay.