geeksville/Micro-RTSP #19: How to stream to YouTube?

SunboX, 2019-09-04:
Do you have any sample on how to stream the camera output to YouTube using a ESP32-CAM module?

`rtmp://a.rtmp.youtube.com/live2/[your-secret-key-here]`

Or is it impossible?

radzio, 2019-10-04:
I'm also interested in that :)

bartlomiejcieszkowski, 2019-11-19:
that's not a rtsp, different format, quick googling gave me these results:

https://www.unifore.net/ip-video-surveillance/yoosee-wi-fi-camera-rtsp-stream-to-youtube-live.html
https://www.stephenwagner.com/2016/11/10/rtsp-to-rtmp-ip-camera-to-youtube-live-streaming/
https://stackoverflow.com/questions/4010019/rtsp-to-rtmp-streaming

tl;dr:
esp32-cam -> any pc with ffmpeg -> youtube

as you need to convert to h.264, the esp32 wont have enough performance to do this

SunboX, 2020-05-01:
Without a PC. 😉 Should be possible with ESP32 only, but I don't know how. Should have enough power and it already can stream to a PC.

kjellxvx, 2020-05-03:
Highly interested in this!

aieseres, 2020-10-08:
I'm super interested in this

Adrianahwjsgsj, 2020-12-22:
Did somebody find something? Or is it impossible?

jimthedj65, 2021-03-28:
you can import the camera rtsp to OBS and output to youtube from there

fazaio, 2021-04-02:
Hy dude im interest too 😍 directly from esp32cam as client to youtube live as server 😍

znsoft, 2021-09-13:
is any one solve it ?

ddamianus, 2022-01-15:
Old post but, NOPE guys! RTMP is only protocol, but reciving server need A/V in supported codec i gues for twitch and youtube be the same so h.264 for video and AAC for audio. On ESP32-CAM and others we have only video. But simple they can not send video in h.264. This board send only pictures in mjpeg copresion. Cheapest solution is rpi zero ;)

arthurd2, 2022-01-17:
Another solution for that would be: use VLC to bridge the ESP32-CAM to the Youtube.

OPinedaCoink, 2022-12-05:
If you use a web stream instead a rtsp stream you can send this stream to youtube
https://www.youtube.com/watch?v=l_j71Dzhlx0&ab_channel=TheHackLife

US4LKW, 2023-02-06:
This Arduino code for ESP-32 CAM, sends the video to the specified server:
```
#include <WiFi.h>
#include <WiFiUdp.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "esp_camera.h"

const char * networkName = "SSID Wi-Fi";
const char * networkPswd = "PASSWORD Wi-Fi";

const char * udpAddress = "server IP-address";  // <= Here I entered the address of my VPS server on the Internet
const int udpPort = 20001; // <= Here UDP port on which PHP script listens

int counter = 1;

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void setupCamera()
{  
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  
  //
  // WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
  //            Ensure ESP32 Wrover Module or other board with PSRAM is selected
  //            Partial images will be transmitted if image exceeds buffer size
  //   
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(psramFound()){  //是否有PSRAM(Psuedo SRAM)記憶體IC
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
[... 197 lines cut ...]
               $color = imagecolorallocate($im, 255, 255, 255);
               imagestring($im, 5, 10, 20, "fps:".$fps." ".imagesx($im)."x".imagesy($im)." ".date("d.m.Y H:i:s")." ".$text, $color);
               imagejpeg($im);
               imagedestroy($im);
               echo '--', $boundary, PHP_EOL;
               $image = "";
               $text = "";
               $frlen = 0;
           }                     
       }
  }
} finally {
  socket_close($socket);
}
?>
```

When I turned on the ESP-32 CAM and then opened the page with the PHP script on my server, in the browser I saw an image from my ESP-32 CAM camera. I still didn't believe it worked. The author of the codes claims that the ESP-32 CAM works like an RSTP client that streams to the server.
Now, just like you, I am looking for a way to stream video from my ESP-32 CAM to my server, because I do not have a permanent IP address for the ESP-32 CAM, so I am interested in the camera transmitting the stream to the server, because the server can't pick up  the stream from the camera. Now I want to stream to the NGNX server, because it has a module for working with RTMP, I have already configured the server, I want to try streaming from this camera to my RTMP server.
Video as work from the author of the scripts:
https://www.youtube.com/watch?v=-BvSkXWC-EE
I express special gratitude to the author of the video.
He really did an incredible thing, he just took and streamed the video to the server!
I just shared this, maybe it will be useful to someone.
I hope that maybe someone will be able to adapt this script in order to send an RTMP stream to the NGNIX server or to send an RTSP stream.