geeksville/Micro-RTSP #32: Esp32-cam as a virtual webcam using ffmpeg v4l2loopback

EderNavarro, 2020-07-10:
**Hi, first of all, excellent work on the RTSP Library for the Esp32-cam.**

I'm trying to use one of these wifi cameras as a virtual webcam,and for now the simplest option is to get the RTSP stream from the camera, use ffmpeg and v4l2loopback to create a virtual **/dev/video1** device.

The module works great, I can see the RTSP feed in VLC using the address **rtsp://192.168.1.201:8554/mjpeg/1** and with the :network-caching=0 in VLC it works like a charm.

but when I use ffmpeg to send the data through the virtual device it only works for 29 seconds

this is the command that i'm using:

**ffmpeg -i rtsp://192.168.1.201:8554/mjpeg/1 -fflags low_delay -fflags nobuffer -vcodec rawvideo -pix_fmt yuv420p -f v4l2 /dev/video1**

and these is the last message before stop working
**frame=  288 fps= 10 q=-0.0 Lsize=N/A time=00:00:29.08 bitrate=N/A speed=1.05x**

I've tried with different camera resolutions but is always the same result. it dies at 29 seconds and speed =1.05

is there any way to make this work??

thanks.

V45370, 2020-11-02:
Hi @EderNavarro ,
I am using this command: 
**ffmpeg  -rtsp_transport tcp -i "rtsp://192.168.1.11:8554/mjpeg/1" -f segment -segment_time 60 -segment_format mp4  -reset_timestamps 1 -strftime 1 -c copy -map 0 camera1-%Y%m%d-%H%M%S.mp4**

I have some broken stream recordings though. I read somewhere in other repos that this is due to some broken jpeg frames. If I found out a solution will let you know.

ManikandanVaradharaj, 2024-11-18:
hai,i need to work on rtsp client in esp32-p4 module..apart from logic i'm not getting proper libraries and components for that module..firstly i need to set lvgl ui support after giving the ip address it should start live stream.for this how could i make the proper way.