geeksville/Micro-RTSP #35: Can this program run on Esp-IDF  environment？

DanielFujiwara, 2020-09-21:
I downloaded your file(Micro RTSP) on GitHub recently.And I had some problems during using it.
Here is the problems:
	1.Your program is base on Arduino IDE. Can this program run on Esp-IDF  environment? 
	2.Will you write a ESP IDF version in the future?
I sent a email to you,too.
Thank you.

brainrecall, 2021-06-18:
I just created an esp-idf example. It can be found here: https://github.com/brainrecall/Micro-RTSP/tree/master/examples/esp-idf

ameerhazo, 2021-06-18:
@brainrecall wow cool! I'll take a look into it. Btw this original Micro-RTSP is written in C++ right? The ESP-IDF is in C and if i remember correctly, if you want to use C++ code in ESP-IDF, you would need to change main.c to main.cpp and write extern "C" for app_main() for it to build. This what I had to do back when I used the Arduino code (C++) in the ESP-IDF environment. Great job!

brainrecall, 2021-06-18:
Thanks! I started with the example from `esp-idf/examples/wifi/getting_started/station` as the base. The usage of Micro-RTSP I based on the POSIX-compliant code from this libraries test at `test/RTSPTestServer.cpp`. I did discover a bug in esp-idf where `SO_RCVTIMEO` with a timeout of zero is actually blocking, so I had to change  `tcpsocketread()` to use `select()` instead. I then bolted together a few other examples for MQTT and OTA updates. I also pulled in performance updates from the `esp-idf/examples/wifi/iperf` example which is greatly missing from the Arduino framework.

I should note, though, the example I wrote will only work with my branch. I should give credit to @kadavris who rewrote the RTSP command sequence handling that greatly improves the connection handshake. I made several other changes to his work to improve buffer handling and actually removing the multiple-client support as that was broken (couldn't handle multiple UDP clients, for one). However, running multiple tasks like I do in the  esp-idf example allows multiple clients to work easily.