geeksville/Micro-RTSP #40: RTSP server on ESP32 as H.264 format for Echo Show

vipt26, 2020-11-20:
Hi, thanks for this but of code to get it as RTSP format.  I'm trying to integrate this via monoclecam to allow the echo show to display the stream but I think it needs to be H.264 format.  Is there anyway to get a version that sends in H.264 format? Thanks

kennethgomez01, 2020-11-26:
For this you don't have to use rtsp, use web streaming instead that you can get live feed from the iframe.

vipt26, 2020-11-27:
> For this you don't have to use rtsp, use web streaming instead that you can get live feed from the iframe.

Hi, thanks for your reply.  I can see the stream with rtsp if i open firefox on the echo show and enter the ip address (haven't tried with normal web streaming yet).  But I've want the echo show to display the stream when i ask alexa.  I've come across monoclcecam which may help but i haven't got round to downloading and installing the gateway yet but some post suggest there may be issues.  So some things i read were that it needs to be rtsp and in h264 format.  Not sure if things have changed.

By the way, what is iframe?

kennethgomez01, 2020-11-27:
Iframe is usually define where an html element use to display image from source at which from the original code from arduino that allow to stream via browser, using jpeg frames.

kennethgomez01, 2020-11-27:
My only solution to your problem is use vlc to convert rtsp to http:// h264

vipt26, 2020-11-27:
hi, can you do that on a live stream so it streams it back out as h.264?

kennethgomez01, 2020-11-28:
Yes, for more reliable and save electric bills I recommend raspberry pi, Id recommend pi 3

vipt26, 2020-11-30:
Thanks for your help, i had a quick google search on my phone how to do this but didn't find much relevance, i'll have a better look towards the weekend.  If you know what it is i'm looking for or a link to a guide or tutorial then please post a link to get me started.

Thanks again for your help.

kennethgomez01, 2020-11-30:
The easiest way to do is to upload the example from library in arduino name camerawebserver after that click start streaming on the browser and right click on the video stream to get the http for your application. I am not 100 percent sure this would work ad your application as this is a low cost device.

kennethgomez01, 2020-11-30:
But if not I will keep you updated to test this thoughts by testing it bymyself

vipt26, 2020-11-30:
Hi, sorry for the confusion.  I can get the webserver running and I can access this using the ip address from my network.  However, I am trying to integrate with alexa.  So I am using the rtsp code on the ESP32 and then trying to direct it through the monocle gateway so that I can ask alexa to show my camera on the echo show.  But its not working, and seems some trouble shooting suggest it needs to be h264 format, but ESP32 is transmitting MJPEG.  I can see the webstream on the echo show if i open the ip address on firefox, but when I ask alexa to display it just says camera isn't responding.  I have a couple of things to try on the monocle gateway this weekend but i don't know if it will work without it being h264.  

I raised this in the monocle forum, here is the link to it:
[https://forum.monoclecam.com/topic/878/esp32-micro-rtsp-on-echo-show-5/5]

Thanks

kennethgomez01, 2020-12-01:
Oh ok, sorry can't help you further. But try the vlc method to convert rtsp to the desire encode.