geeksville/Micro-RTSP #46: Stream to ip address

filzek, 2022-03-11:
Hi,
How could we stream from esp32 to a ip address?

Like I want the esp32 cam to stream to  myfeedvideo.io

How could IRTSP to a particular address?

V45370, 2022-03-11:
I used https://github.com/easytarget/esp32-cam-webserver to do this. You need a domain to point to your static ip address. If you use a router you need port forwarding to <ESP32-CAM-IP-ADDRESS>:<PORT(default is 81)> and you are done.

filzek, 2022-03-11:
> need port forwarding to :<PORT(default is 81)> and you are done.

I took a read on https://github.com/easytarget/esp32-cam-webserver  but I
don't find where it it show how to send the video to an IP it doesn't have
that option there.

What I mean is not to a externa cliente connect to the esp32 but that esp32
create a video broadcast to an external IP.

Like you program esp32 to broadcast the video a cliente  ip:port, to if you
open the application in the wild (internet) you will get the video feed.

Did you understand ?

Bra1nK, 2022-03-11:
That's not how streaming works, you receive not send, perhaps if you explained what you were trying to achieve.

V45370, 2022-03-11:
If only 1 person is going to view the stream, you can do it as I explained above. Right now I am viewing my stream with mydomain.com:81/view. However if you need to stream to more people I guess you could use ffmpeg from another PC in the network to upload to youtube for example. This https://gist.github.com/olasd/9841772 seems promising but I haven't tested it.

filzek, 2022-03-11:
So, do you think its possible to do it?

Like an app open in the smartphone over 4g or wifi it do a websocket to
keep the IP and PORT open for external access, send to the cloud its IP
address and a request.
The device receive the mqtt or a get/post to the cloud and receive the IP
address and video feedr equest.
the device start to broadcast to the IP address and port of the smartphone.

so its quite simple and works under any condition.

 *Erick MacDonald Filzek *
Phone / WhatsApp: +55 (19) 981416414

Bra1nK, 2022-03-11:
It probably varies with mobile phone providers but mine gives you a private ip address (10.0.0.0/8 range) which is normally not routable so you can't run a server reachable from the Internet. Like I said it usually works the other way round, the RTSP stream is the server and the mobile client connects to it.

garudaonekh, 2022-09-14:
Hello, I am also finding solution for this---stream to a public IP.

For example, I can setup an RTSP server and let the ESP32-Cam to stream to that server.

wiwidnadw, 2022-10-04:
Here i have project example to stream (esp32cam) into ip address. So this one called as IPCamera, but i using other interface (VB.net). Hope it refresh viewpoint

https://youtu.be/mogghaQbEy8

filzek, 2022-10-05:
This is not what the subject are asking ti.

You are posting a request to the esp32, what we want is that a esp32 send
the images to an IP address and port without receiving any request!