geeksville/Micro-RTSP #54: Compiling error in Arduino IDE

orlin369, 2024-02-03:
Hello I am using Arduino IDE and trying to build the project.

This error comes out:

```
C:\Users\User\Documents\Arduino\libraries\Micro-RTSP\examples\ESP32-devcam\ESP32-devcam.ino: In function 'void loop()':
C:\Users\User\Documents\Arduino\libraries\Micro-RTSP\examples\ESP32-devcam\ESP32-devcam.ino:220:40: error: no matching function for call to 'CStreamer::addSession(WiFiClient&)'
         streamer->addSession(rtspClient);
                                        ^
In file included from c:\Users\User\Documents\Arduino\libraries\Micro-RTSP\src/SimStreamer.h:4,
                 from C:\Users\User\Documents\Arduino\libraries\Micro-RTSP\examples\ESP32-devcam\ESP32-devcam.ino:6:
c:\Users\User\Documents\Arduino\libraries\Micro-RTSP\src/CStreamer.h:16:19: note: candidate: 'CRtspSession* CStreamer::addSession(SOCKET)'
     CRtspSession *addSession( SOCKET aClient );
                   ^~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\Micro-RTSP\src/CStreamer.h:16:19: note:   no known conversion for argument 1 from 'WiFiClient' to 'SOCKET' {aka 'WiFiClient*'}

exit status 1

Compilation error: no matching function for call to 'CStreamer::addSession(WiFiClient&)'
```

Would you be able to help?

Regards

lejibxl, 2024-02-12:
I have the same error

canDry, 2024-02-23:
Using this library:  https://github.com/circuitrocks/ESP32-RTSP/tree/master  worked for me.  
Note: remember to first remove the existing Micro-RTSP from libraries before importing this one.

liuyang3688, 2024-05-06:
class constructor or function parameter need a pointer instead of object.

repo code has obvious error.

gajarskymartin, 2024-12-05:
for this particular issue it is enough to put an address to addSession
```streamer->addSession(&rtspClient);```